makerbase-mks/MKS-Monster8 #5: Unable get mcu id

ddpsft, 2021-12-08:
I'm trying to get my mcu id to reach my board using RatOS/Mainsail/Klipper. I'm using the provided bin and cfg files, following klipper's documentation and trying any id I found here, but only having these answers:

18:11:42
Printer is not ready
The klippy host software is attempting to connect. Please retry in a few moments.

18:21:03
mcu 'mcu': Unable to connect
Once the underlying issue is corrected, use the "FIRMWARE_RESTART" command to reset the firmware, reload the config, and restart the host software.
Error configuring printer

Any ideas? I used the same RatOS with a "supported" board and that was much easier.

Klipper and Mooraker log files:
[klippy.log](https://github.com/makerbase-mks/MKS-Monster8/files/7679045/klippy.log)
[moonraker.log](https://github.com/makerbase-mks/MKS-Monster8/files/7679046/moonraker.log)

mks-viva, 2021-12-09:
What you use software and how to get mcu id?

ddpsft, 2021-12-09:
I'm using the provided bin and cfg files for the board and the latest ratOS for the rpi3B+. To get the id I tried to use ls /dev/serial/by-id/* and by-path.
I was using it without a display, just connecting the board to the pi and now I'm trying to do it with connected display to understand what is going in in the board. The first time I updated it was with no display, thus I don't know what happened and now it stucks with "Booting" text for a long time. Then I tryed to flash the bin file again and now it is stuck with "updating 100%" instead of rebooting. My other board (Robin Nano v3) do it and reboots normally, but not the monster8

ddpsft, 2021-12-09:
Fixed that using Putty and SSHing the board, but now I can't go further because apparently the provided bin file is using a old version of klipper (host=v0.10.0-122-gc179db3d, mcu=v0.9.1-580-g478f26ca)

mks-viva, 2021-12-11:
Get MCU id, use 'ls /dev/serial/by-id' command by putty.exe
[image]

ddpsft, 2021-12-14:
Done. Thank you for your help.

Instead of opening another issue, I'll add one more question. If it's the wrong way, please close the issue and I'll be opening another using the right way.
Is there any documentation about how to edit config.cfg in order to use TS35 display? Monster8's page states that it is compatible, but i've found another issue where was discussed that there is no support. I'm trying do start from the provided voron cfg file (using mini12864) and so far it still showing only "Booting..." on TS35

closed 2021-12-16 by ddpsft