makerbase-mks/MKS-Monster8 #6: SPI interface and TS35 display

ddpsft, 2021-12-16:
Is there any documentation about how to edit config.cfg in order to use TS35 display? Monster8's page [here](https://github.com/makerbase-mks/MKS-Monster8) and in [Aliexpress](https://www.aliexpress.com/item/1005003292898244.html) states that it is compatible, but i've found another issue where was discussed that there is no support. I've used the code from generic klipper code and now I'm trying to adapt the provided voron cfg file (using mini12864) and so far it still showing only "Booting..." on TS35.

Additionally, where can I found any detail about pins on SPI3 interface? I'd like to use my adxl345 chip connected to the board instead of connecting to RPi's GPIO pins. The [provided pin "map"](https://github.com/makerbase-mks/MKS-Monster8/blob/main/hardware/MKS%20Monster8%20V1.0_002/MKS%20Monster8%20V1.0_002%20PIN.pdf) doesn't detail it thus I'm no able to identify VCC, G, CS, SDO, SDA and SCL pins.

mks-viva, 2021-12-17:
If you use MKS TS35 for Monster8, you only use marlin 12864 UI or marlin color UI

closed 2022-01-11 by mks-viva

rafaelnunesdomingues, 2022-07-23:
I have a monster 8 running on a voron 2.4 with marlin. I'm using TS35 which connects to EXP1 and EXP2. How can I implement a Sdcard reader?