makerbase-mks/MKS-Monster8 #7: Mks Monster8 board firmware flash problem

turnali, 2022-01-31:
Hello, 

I just received my monster8 board that I purchased from Aliexpress store. I have been trying to flash the firmware for the klipper for 2 hours but the board doesn't take the firmware. I have even tried marlin firmware but still the board doest take the ''bin'' file. I also downloaded the sample ''firmware.bin'' files from the mks github page to see if those work but still no luck. The board doesnt flash neither klipper nor marlin firmware. 
I tried 6 different SD cards( 3 different 128mb,1gb, 8gb, 16gb) to flash but none of them worked.  I am using MKS TMC2209 drivers.
Can you help me please?

DimasVl, 2022-02-01:
What display are you using? What signs indicate that the firmware is not working?

turnali, 2022-02-01:
I am using MKS MINI12864 V3 because of the klipper firmware. 
After I flash the klipper firmware, the board never connects to the raspberry pi. And after I flash the marlin firmware, the board never connects to the display.
But most importantly, after I flash the firmwares, I checked the SD cards if the BIN files turned to CUR files but they never do. Everytime they stay as BIN files

turnali, 2022-02-01:
By the way, I also tried BTT TFT35 V3 display but still no luck

DimasVl, 2022-02-01:
Maybe you should try this method? https://github.com/makerbase-mks/MKS-Monster8/wiki/DFU_upload

DimasVl, 2022-02-01:
> Can it be a faulty board?

I don't think the board is faulty. Maybe it's just a bootloader problem and you need to repair it. In this board, this is very easy to do with a usb cable.

DimasVl, 2022-02-01:
I bought Mks Monster8 board and ts_35 display from Aliexpress store a few days ago. First, I configured the marlin firmware with a ts_35 display, uploaded the firmware to a usb flash card. I did not connect anything to the board, even drivers and a display. Then I connected the USB flash with the firmware file, applied power to the board, waited a few minutes and turned off the power. After that, the extension of the mks_monster8.bin file changed (I don’t remember the name of the extension). Then I connected the display and saw the marlin menu on it.
I had a problem with usb and TF card - I could upload the firmware via usb, but on the marlin screen, a usb error was written and there were no g-code files. I spent 2 days to find a setting in marlin that fixed this problem. This setting was not mentioned in the Mks Monster8 manual, I had to spend a lot of time to find it.
Now the board is connected to the printer, 3D Touch is connected, with the TMC2209 drivers, the motors have become silent. I'm very happy ). Here is my short story.)
Maybe I'm not expressing myself clearly - but English is a foreign language for me and I use Google translator.

DimasVl, 2022-02-01:
> I am using MKS MINI12864 V3 because of the klipper firmware. After I flash the klipper firmware, the board never connects to the raspberry pi. And after I flash the marlin firmware, the board never [...]

I forgot to ask which slot you are using for firmware - on the LCD screen or on the board? Do not use the LCD screen's SD slot for flashing. I would advise you to try using a usb flash card and a usb connector on the board. Try uploading the firmware from a USB flash drive, then check if the extension of the firmware file has changed or not. If the extension changes, then you will only need to configure the firmware for your display in order to see the image. I can compile marlin firmware for Monster8 + MINI12864 V3 to make sure the hardware is working.

turnali, 2022-02-01:
> I bought Mks Monster8 board and ts_35 display from Aliexpress store a few days ago. First, I configured the marlin firmware with a ts_35 display, uploaded the firmware to a usb flash card. I did not [...]

Hi, 
Your advise about the uploading the firmware via usb flash drive without connecting anything really worked. Its very weird but This should not be the way to flash a firmware very time I need to. Thanks alot about that anyways. 
I was able to upload the firmware and the file extension changed to CUR. But I still cannot connect to raspberry pi. 
I am using Manisailos and klipper firmware. I am also now using the BTT TFT35 V3 display with this setup. 
The board doesnt give serial number from klipper which I using Putty for it but I copied the serial from the mks generic printer.cfg file.
I have tried everything I could to connect the board to the raspbery pi for last 3 hours but although I uploaded the klipper firmware to the board, it just doesnt connect to the Pi. 
I am using the type C Usb cable that came with the Monster8 board to connect to the Pi.
From the Mainsail interface, it keeps giving ''mcu'' error. 
I just gave up trying now. I am also afraid of damaging the board everytime I unplug all the connections and drivers to upload a firmware trying to fix this problem

turnali, 2022-02-01:
> I forgot to ask which slot you are using for firmware - on the LCD screen or on the board? Do not use the LCD screen's SD slot for flashing. I would advise you to try using a usb flash card and a [...]

I was ofcourse using the SD slot on the board not the display. I am not an amateur

DimasVl, 2022-02-02:
> I have tried everything I could to connect the board to the raspbery pi for last 3 hours but although I uploaded the klipper firmware to the board, it just doesnt connect to the Pi.

I've never used klipper and can't offer advice. Pay attention to the klipper configuration, especially the availability of the com port. In the marlin firmware provided by the board manufacturer, the com port is disabled. When it becomes necessary to transfer data to third-party devices, it is necessary to include a com port in the firmware.

turnali, 2022-02-05:
I installed a bootloader to the board. Now the board takes firmware from the sd slot without unplugging anything. And it also gives serial number to connect to raspberry pi and klipper. After lots of hassle, it finally works.
Its interesting that why the board didnt come with a bootloader installed from the factory. Its the first time I see something like this.

makerbase-mks, 2022-02-14:
@turnali Have you change the name of the firmware to be "mks_monster8.bin" before upload?

TrulyMarcus, 2022-10-29:
Can someone please tell me how to upload firmware (or maybe a bootloader) via USB flash drive? My board is no longer noticed by my computer's USB ports.

tranhuunhan82, 2022-12-08:
You can try:
1. Copy firmware .bin to root of SDCard
2. Unplug power of MKS Monster 8
3. Press Boot0 in board
4. Insert SDCard to board
5. Plug Power of board
6. Waitting... to complete

Good luck.

michelspas, 2022-12-16:
On my side it only worked after renaming  Firmware.bin to  mks_monster8.bin.
Then it was OK.
Now I am having problem with the usb to usb-c cable that came with the board.  I am loosing connection.  The printer is stable with my son's similar cable. 

Can we use regular USB to regular USB wire to connect to that board?

Dawaitizova, 2023-01-12:
"I had a problem with usb and TF card - I could upload the firmware via usb, but on the marlin screen, a usb error was written and there were no g-code files. I spent 2 days to find a setting in marlin that fixed this problem."

@DimasVl what setting did you use to fix the USB TF card.

marcoslajovic, 2023-08-12:
I managed to put the firmware by doing the following procedure:

1) I connected the MKS Monster board to the PC USB port. The board then appeared in the Windows Device Manager on the COM3 port.
3) I pressed the boot0 and reset buttons on the board at the same time and released it.
4) In the Windows device manager, the board disappeared from COM3 and appeared in Universal Serial Bus Devices under the name of STM Bootloader.
5) I opened the command prompt and entered in the DFU program folder (taken from MKS Monster's github)

cd Users\marco\Downloads\DFU-Upload\DFU-Upload

6) I entered the following command to upload the firmware

dfu-util.exe -a 0 -s 0x0800C000:leave -D mks_monster8.bin -d 0483:df11 dfu-util 0.8

7) The upload was done.

TheMajster007, 2023-10-31:
> Maybe you should try this method? https://github.com/makerbase-mks/MKS-Monster8/wiki/DFU_upload https://github.com/makerbase-mks/MKS-Monster8/tree/main/tool/DFU-Upload

It helped to install the drivers from widnows via zadig.exe, then I ran DFU-Upload_Bootlader.bat. Once I had that installed I copied the firmware.bin file from the raspberry pi and installed it via DFU-Upload-firmware.bat

alucard321pl, 2025-08-21:
I know this is an old issue, but what worked for me was renaming the firmware file to mks_monster8.bin and flashing it from a USB pendrive. I had previously tried several different SD cards, and none of them worked.  

It seems like some boards are picky about the storage medium — even though the SD cards were formatted correctly and worked elsewhere, the board just wouldn't recognize them. Using a USB stick solved the problem instantly. Might be worth trying if you're stuck.