makerbase-mks/MKS-Monster8 #11: updating firmware - nothing works

Benster1979, 2022-04-15:
I've tried the following to get it to upload a firmware

SD card - doesn't convert to *.CUR (did it once then on reboot, no longer available)
USB Stick - doesn't convert to *.CUR
DFU-Util - Partially

DFU-Util doesn't appear to work with Windows 10 or 11. Tried on various machines.

Did get DFU-Util on pi to upload both the bootloader and the firmware but it still isn't being detected by the pi outside of DFU mode. "ls /dev/serial/by-id/*" via terminal shows no directories. Doesn't matter if I do make menuconfig or not.

I've tried unplugging screen, drivers and removing the UART jumpers

It never detects the board outside of DFU mode. I've also tried loading the klipper.bin via DFU but still no serial device detected.

Thought I might try your Marlin 2x but it doesn't compile due to errors in features/z-align

Please help

MKS-hosiey, 2022-04-19:
It should be possible to update the firmware with a TF card. Have you checked that your TF card is normal? DFU can work on win10; whether the motherboard is powered by usb to update firmware or switch power supply

closed 2022-07-11 by makerbase-mks

MarqSolucoes, 2024-06-21:
I'm having the same problem, what was the solution?