makerbase-mks/MKS-Monster8 #12: CANbus usage ...

themacboy, 2022-04-19:
It is possible to connect a expansion board like FLY SHT36 board in the Monster8 CANbus port and control this expansion board? 

If possible, Can you give a few instructions about how to do it?

Thx in advance !

710920772, 2022-06-06:
hello, you can use MKS USB to Can, And using Usb to connect the board.

tungyao, 2022-08-27:
1. main board need to select `USB to CAN bus bridge` on communication interface optioin
2. SHT board select `CAN bus` option
3.  
[image]

4. `IMPORTANT` If cannot connect the main board , try to manually set baud rate `ip link set can0 up type can bitrate 500000`