makerbase-mks/MKS-Monster8 #17: Configuration for use MKS H43 on board Monster 8 V2

LoloMake60, 2022-07-21:
Hi,

I would like to connect my MKS H43 screen to the monster 8 V2.

What is the configuration to use USART1?

For Marlin firmware.

LoloMake60, 2022-07-21:
[image]

MKS-hosiey, 2022-07-22:
#if HAS_DGUS_LCD
  #define LCD_SERIAL_PORT 1
  #define LCD_BAUDRATE 115200

Hello, you only need to configure the serial port to 1 in the advanced configuration, and the others are configured according to H43

LoloMake60, 2022-07-22:
Hi,

Thank you perfect! It works!

I add #define LCD_SERIAL_PORT 1 in the advanced configuration.
[image]

It works for screen e3v2.
Add line under in pins_MKS_MONSTER8_common.h
[image]

Thank!