makerbase-mks/MKS-Monster8 #18: Configuration for use MKS H43 on board Monster 8 V1

satinfo50, 2022-08-06:
Hello to all
I bought a mks-monster8 version one board and a mks-h43 LCD some time ago
Unfortunately, I did all kinds of configuration on the files so that they can be connected to each other through uart, but when I turn it on, only the makerbase logo appears on the screen.
Can someone configure the configuration.h and configuration-adv.h files for Marlin-2.1.x so that this display works on the board?
I tried everything but it didn't work.
i attached my configuration to see what i done
[configuration.zip](https://github.com/makerbase-mks/MKS-Monster8/files/9274486/configuration.zip)
.

LoloMake60, 2022-08-06:
Hi,

I invite you to look at this ==> https://github.com/makerbase-mks/MKS-Monster8/issues/17

satinfo50, 2022-08-06:
Unfortunately, there is no such setting in configuration_adv.h in Marlin-2.1.x