makerbase-mks/MKS-Monster8 #19: USART 3 on MKS MONSTER 8 V2

LoloMake60, 2022-08-06:
Hi,

I would like to have confirmation on the rt and tx usart3 pins.
RX = PD8 and TX = PD9.
[image]

Thank you for your confirmation.

mks-viva, 2022-08-16:
PD8 is USART3_TX, PD9 is USART3_RX

mks-viva, 2022-09-03:
Use Klipper firmware?

LoloMake60, 2022-09-03:
no i use marlin

rai0012, 2023-01-31:
Uart3 works for me with marlin

LoloMake60, 2023-02-01:
Hi,

You use usart3 for screen display?

Cordialement,

rai0012, 2023-02-02:
Yes, in Marlin and in Klipper.
What I can't get is the wifi to work in Marlin.

LoloMake60, 2023-02-03:
Hi,

Maybe that's it.
[image]
For me it's works with SERIAL_PORT_1 -1 and SERIAL_PORT_2 1.

rai0012, 2023-02-03:
But the wifi works for you

LoloMake60, 2023-02-07:
Yes the wifi works but the driver 7 E4 it's disable because the jumper move pd11 to wifi.
[image]
I have flash my mks wifi with ESP3D.
But my screen e3v2 not works, RX and TX on PD9 and PD8. 
I don't understand!

rai0012, 2023-02-11:
Let's clarify a bit.
I use marlin I have the dgus screen connected to uart 3 and working, I have the wifi module connected in its slot and it already works for me with flashed esp3d.