makerbase-mks/MKS-Monster8 #25: mks robin wifi

mozg4D, 2022-12-23:
failed to make mks robin wifi to work with mks monster 8
failde to make MKS TS35 V2 to work in TFT_LVGL_UI mode
need manual

mozg4D, 2022-12-24:
Seems like: in order to run wifi, one need tft screen in LVGL UI mode.
The only supported tft screen is TS35, which requires SPI flash to run in LVGL mode, which is missing
So.. is wifi (mks cura plugin) impossible on monster8?

marcoslajovic, 2023-08-12:
I would also like to make the WIFI module work on the monster board. I thought about the possibility of connecting a TFT display that has support for the WIFI module on the Monster, but unfortunately the board does not have the AUX1 connector, only the EXP1 and EXP2 connectors. Is it possible to make a TFT touch display work on the Monster?

ellensp, 2026-09-08:
Please see https://github.com/MarlinFirmware/Marlin/pull/28562   A work in progress

ellensp, 2026-09-19:
Now tested and working on at least two monster8 v2's

It needs to be tested on the V1 boards, as I suspect it will use the same bootloader that enables the USB host and disables the mks wifi module baud rate switching and reboot control. 

ie it may need this moved to pins_MKS_MONSTER8_common.h

```cpp
/**
 * The MKS bootloader hands over with USB OTG_HS still clocked and its embedded
 * PHY powered. That PHY drives the OTG_HS_DM / OTG_HS_DP pads (PB14 / PB15)
 * directly and overrides the GPIO output driver, so those two pins cannot be
 * driven at all - Marlin sets MODER and ODR and the pins never move. Release
 * it before anything else runs, unless this build really uses OTG_HS.
 *
 * Power the PHY down before stopping the clock; the register is unreachable
 * once the peripheral is unclocked.
 */
#if !defined(USE_USB_HS) && !defined(USE_USB_HS_IN_FS)
  #define BOARD_PREINIT() do{ \
    if (TEST(RCC->AHB1ENR, RCC_AHB1ENR_OTGHSEN_Pos)) { \
      *(volatile uint32_t *)(USB_OTG_HS_PERIPH_BASE + 0x38) &= ~USB_OTG_GCCFG_PWRDWN; \
      __HAL_RCC_USB_OTG_HS_CLK_DISABLE(); \
    } \
  }while(0)
#endif

```