makerbase-mks/MKS-Monster8 #26: V2 board starts burning when attempting dfu mode

boannarrow, 2022-12-24:
My monster8 v2 starts burning on the marked piece after holding the boot button while the board is off and turning it on with the button pressed. I'm trying to add marlin binary file transfer via octoprint. Marlin 2.1-bugfix. My ras-pi and my computer see my board as a com port but the .bat files don't run and dfu-utils isn't recognizing my device.
[image]

MKS-hosiey, 2022-12-29:
you can  copy the mks_monster8.bin file to the TF card, and you can update it with the card, or update the .bin file through dfu

casiobearing, 2023-01-23:
I would like to confirm this "V2 board burning" issue - What bonnarrow means is that when you put the Monster8 board into DFU mode, HE-0 is enabled (the red LED turns on), causing Extruder0 to begin heating. There could cause a hot end fire if the board remains in DFU mode for an extended period of time.

casiobearing, 2023-02-02:
In the end I think Q7 MOSFET was a faulty component from manufacturer. Not good ;O(
Got naff all help from the Monster8 supplier so I repair the board myself by replacing Q7 MOSFET with a equivalent one and also changed MOSFET drive signal from PB10 to PA8. This just needed removing of the 100Ohm SMD resistor situated near the external drive jumper.
I'm going to be using a 240VAC mains powered bed heater anyway which is way more efficient and this will take the strain off the bed heater circuitry - using high power MOSFETs for this application isn't a good idea.

Anyway the repair is working and entering DFU mode doesn't cause problems any more.

FauthD, 2023-07-13:
Well, I do not think this is a faulty device. To much coincidence for this many cases.

For Q7 the situation in DFU is the same as in normal bed heating (if 3.3V is applied from the TX3).
Perhaps the ouput of TX3 (PB10) is much less (1.8V ?), leading to a partially switch on situation that heats up Q7 until it dies.

Just my guess after 50 years of experience in electronics.

masapod2892, 2024-04-09:
Just encountered the same issue on the v1 board but in a different scenario. When I was heating the bed up to do my first print I smelled something burning then to my surprise the mosfet was hot enough to desolder itself.