makerbase-mks/MKS-Monster8 #32: How to - Visual Code, STM32CubeProgrammer, compiling Marlin and upload via USB C DFU Boot0 method

casiobearing, 2023-01-21:
Here's how to upload firmware to Monster8 using STM32CubeProgrammer via the USB C connection.

Install Visual Code and then add in the extensions for PlatformIO and "Auto Build Marlin".
Download the Marlin source code zip file from github and unzip it somewhere accessible for Visual Code.
Now find in that unzipped code the file in folder "Monster8\Marlin-2.0.x\ini" called "stm32f4.ini"
You want to edit that file to change lines 410, 411 to read as follows -->
board_build.offset          = 0x0000
board_upload.offset_address = 0x08000000

Now use Visual Code to compile (using Marlin AutoBuilder works fine for this) using the environment called "mks_Monster8".
The compiled file will be found in folder "Monster8\Marlin-2.0.x\.pio\build\mks_monster8" and it will be
called either "firmware.bin", "firmware.elf" or "mks_monster8.bin"  (you can use either of them for the following download process).

Attach your Monster8 board to your computer via its USB C socket (you can't use the USB 2.0 connector for this activity).
Press and hold down the "Boot0" button and either power on the Monster8 board or press and release the "Reset" button.
Your computer should then detect a USB device has arrived - It will be show right at the bottom of Windows "Device Managers list", called "STM32 BOOTLOADER".

Run up STM32CubeProgrammer, in the top right corner, select "USB" and click "Connect". This will connect to the Monster8 board and show details about the STM32F407 processor. In STM32CubeProgrammer, Open File to the "firmware.bin" file from above, and then click "Download" to send it to the Monster8.
Then just reset or reboot your Monster8 board to get your new code up and running.
Hope this helps anyone stuck trying to get things going :o)

casiobearing, 2023-01-28:
I should warn you that if you use decide to use BOOT0 DFU mode, that you should physically disable your heat bed (e.g. by removing fuse FS1) before entering DFU mode. See issue #26 if in doubt why.
I have modified my Monster8 board to change the MOSFET Q7 gate drive signal. I disconnected PB10 (by removing the 100 Ohm resistor) and connected the MOSFET gate to PA8 (used for 3DTouch). This modification stops the hotbed from heating up when you go into DFU mode.

closed 2023-01-28 by casiobearing

gitcnd, 2024-12-06:
For the curious, I do not think this works on the Monst8 V2 (I guess they are using emulated USB, and not native serial now?):

Windows no longer detects anything when the board goes into DFU mode.

The SWD pins do work to load and backup firmware using j-link at least.

This pinout is confirmed working for that:
[image]

HattonLe, 2024-12-06:
I use a real USB port on my Dell M6700, and Windows running baremetal i.e. no VM.