makerbase-mks/MKS-Monster8 #41: Using MKS TFT35 display in MKS Monster8 V2

marcoslajovic, 2023-08-13:
I've been looking for the last few days how to use a MKS TFT35 touch screen display on the MKS Monster8 board without success. I finally figured out how to do it so I decided to create this post.

On the MKS TFT35 display, connect a 4-pin cable to the AUX-1 connector on the 4 lower pins where it says TX RX GND +5V.

[image]

On the Monster8 board, connect the same cable with the same sequence to the UART connector (next to the CAN I2C connectors). In this connector, the pins are in the sequence from top to bottom like this: RX1, TX1, GND and +5V.

[image]

The TX pin from TFT35 display must be connected to the RX1 pin in monster board, the RX pin in TFT35 display must be connected to the TX1 pin in TFT35 board.

In Marlin 2.0 firmware in config.h set serial_port2 equal to 1:

[image]

In the variant.h file that is in the folder: buildroot/share/PlatformIO/variants/MARLIN_F407VE
Change the serial_uart_instance to 2:

[image]

With this change, the UART connector on the board becomes Serial2 (USART2), which will connect to the Serial_port_2 configured in config.h. Serial_Port_2 receives the value 1 to say that it is the first serial port on the board, although it already has a serial port but which is defined by default at -1 to make the connection with the PC.

The following photo shows the system working with the display reading the plate temperature sensor.

[image]

guesch39, 2024-07-06:
Hello,
I did the same combo with the MKS TFT35 V1 on my Monster 8 V2 and it works perfectly. I have the BTT Firmware ( MKS_TFT35_V1_0.27.x_encrypted.bin ) and it works perfectly.
However, when I plug the WIFI module on my lester 8 V2 the connection between the screen and the card does not work.
I know that the wifi module uses uart 1 and the screen too.
How can I resolve this conflict?

LucasCarapella, 2024-10-17:
@guesch39 Hi, were you able to solve this problem?

guesch39, 2024-10-17:
Hi, no the problem is present, I unpluged the MKSWIFI !

LucasCarapella, 2024-11-01:
Hola, @marcoslajovic @guesch39 
¿cuando pongo a calentar el extrusor calienta, pero no se actualiza el valor de la temepratura actual, tuviste este problema?

Es como si la madre recibiera la información de calentar, pero esta no le informa a la pantalla a que temperatura está. Ya modifique y probe valores en el puerto serie, y en la configuración del buffer seri

LucasCarapella, 2024-11-01:
Hello, @marcoslajovic @guesch39 
When I heat up the extruder it heats up, but the current temperature value is not updated. Did you have this problem?

It's as if the mother receives the heating information, but it doesn't inform the screen what temperature it is at. I've already modified and tested values ​​in the serial port, and in the serial buffer configuration.