makerbase-mks/MKS-Monster8 #50: Monster8 v2.0 Neopixel pin(PC5) not working

ivantem86, 2024-11-12:
Hi
Voron的Stealthburner, 使用三顆連接的SK6812, 不點亮
已進行過的排查:
1. 同一串LED接到Arduino測試, 卻可以點亮, 並可依指令變色.
2. 在Stealthburner的空中接頭處使用電表測量, +5v, 且正負極的接線無誤
3. 在靠近主板的data線, 接上100歐姆電阻, 仍不點亮
4. Mainsail控制介面中, 有列出兩個neopixel, 分別是min12864與sb_leds(表示config有被讀入), 前者可點選顏色變色, 後者卻無反應.
5. 確認config中, pin: PC5, 符合官方PIN diagram對此pin的編號. color_order則是RGBW與GRBW都試過了仍無效

請問, 是否應將SK6812的data線拉到其他pin, 或者還有進一步偵錯的手段?
謝謝
Ivan

kwankiu, 2024-12-08:
Same here with WS2812B Neopixel, PC5 doesn't work, as a workaround, I connected it to the 3DTouch Pin PA8 instead.

ivantem86, 2024-12-19:
> Same here with WS2812B Neopixel, PC5 doesn't work, as a workaround, I connected it to the 3DTouch Pin PA8 instead.

Yes, I also found PA8 is alternative, I posted this finding on discord but forgot to reply myself on this github post. thank you for posting the solution.

ellensp, 2026-09-19:
E2-DIAG jumper needs to be removed. From J23  or it connects the PC5 to the diag pin

[image]